git-kick/ioBroker.e3dc-rscp #210: Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7

mibr85, 2024-03-08:
similar to #209 
i have the same issue every 5 min this massage come up:

e3dc-rscp.0 | 2024-03-08 06:15:17.042 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
e3dc-rscp.0 | 2024-03-08 06:10:17.040 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
e3dc-rscp.0 | 2024-03-08 06:05:17.039 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
e3dc-rscp.0 | 2024-03-08 06:00:17.039 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
e3dc-rscp.0 | 2024-03-08 05:55:17.037 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
e3dc-rscp.0 | 2024-03-08 05:50:17.035 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7
e3dc-rscp.0 | 2024-03-08 05:45:17.038 | warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7

After restart of IOB container, problem is not solved.
What mean this warning?

git-kick, 2024-03-08:
Part 1: `warn | Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7`
This looks like your E3/DC device sends a tag which is (to my knowledge) not documented and thus cannot be decoded. This does not affect all other tags and values. I am happy to include the new tag if somebody can figure out what it is.

Part 2: `warn | TAG_EMS_REQ_... has no assigned polling interval - assuming 'M' - please check io-package.json`
I checked the current `io-package.json` and it contains all three listed tags. The warnings are normally due to re-use of an older adapter configuration which did not contain the tags (similar issue #191).
Please create a completely new adapter configuration (_not_ by importing a json) as described in the README under [Reuse of adapter configuration](https://github.com/git-kick/ioBroker.e3dc-rscp?tab=readme-ov-file#reuse-of-adapter-configuration).

mibr85, 2024-03-10:
Is there an option to find out what the Unknown tag: tagCode=0x100003e, len=4, typeCode=0x7 means?

git-kick, 2024-03-10:
`len=4, typeCode=0x7` says it is a `UInt32` value. About the meaning of `tagCode=0x100003e`, the developers at E3/DC know...
There is an official list of tag codes, and also a [tag-index](https://jnk-cons.github.io/easy-rscp/kdoc/easy-rscp/de.jnkconsulting.e3dc.easyrscp.api.frame.tags/index.html) by jnk, but this value is not listed yet.

closed 2024-03-10 by git-kick

git-kick, 2024-05-27:
see v1.2.6, currently in beta repo

flikflakandback, 2024-06-13:
Thank you for all the work you do for us! I tried to install v1.2.6 from the beta repo, but it didn't show up? Am I missing something?

git-kick, 2024-06-13:
Sorry for that, I'm still struggling with the repo process (this time, I missed to update the copyright to 2024...). As I am on travel, I can't fix this short term. If you like, you can install v1.2.6 from github, using the URL https://github.com/git-kick/ioBroker.e3dc-rscp/tree/v1.2.6