git-kick/ioBroker.e3dc-rscp #211: Only warning per unknown tag

git-kick, 2024-03-11:
**Current situation:**
The adapter places a warning every time it receives an unknown tag, which can result in many identical warnings.

**Proposed situation:**
After every instance start, the adapter warns about an unknown tag the first time. Afterwards, warnings are suppressed. 

**See also:**
#210 and others.

git-kick, 2024-03-28:
Fixed for tag 0x100003e in branch [0x100003e](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/0x100003e)

@mibr85 and @ESDN83 this should solve the many warnings issue. If you like to test, please install from the URL above.

Will be released with v1.2.6 (later)

somethingcreator, 2024-03-30:
Hallo, hier bei mir das Gleiche. Danke für den Fix. Wird er dann als reguläres Adapter-Update unter iobroker-GUI verfügbar sein, oder muss man es manuell einspielen?

git-kick, 2024-03-30:
Hallo, der Fix kommt mit v1.2.6, dafür ist aber noch kein Termin gesetzt. Wer den o. g. Branch installiert, hat genau v1.2.5 und den Fix.

git-kick, 2024-05-27:
see v1.2.6, currently in beta repo

mibr85, 2024-06-25:
Hallo hab dir soeben mal die aktuelle RSCP liste per mail geschickt evtl hilft sie dir ja :-)