git-kick/ioBroker.e3dc-rscp #214: Ladeleistung WB_1 steuert WB_0

P-B-aus-R, 2024-04-20:
Hallo,
ich habe eine zweite Wallbox am E3DC installiert. Es sind automatisch die Objekte für WB_1 im ioBroker aufgetaucht. Ich habe einen Fehler festgestellt:
Wenn ich über 
    e3dc-rscp.0.WB.WB_1.Control.PowerLimitation
die Leistung ändere, dann ändert sich die Leistung von WB_0 (EasyConnect) und nicht von WB_1 (MultiConnect).
Wäre schön, wenn das Problem in einer nächsten Version beseitigt werden könnte.
Schon mal Danke dafür.
Peter

git-kick, 2024-04-22:
@ka-vaNu, magst du dir das ansehen?

ka-vaNu, 2024-04-23:
Kann ich gerne machen. Das ist aktuell eine bekannte Einschränkung. Mangels Testmöglichkeit kann ich hier aber nur geringe Hoffnung machen.

P-B-aus-R, 2024-04-23:
Falls es Dir was hilft:
Mit dem Programm go-rscp kann ich mit folgendem Befehl die Ladeleistung der zweiten Wallbox, also WB_1 schalten (die 11 sind die Ampere):

./e3dc '[["WB_REQ_DATA",[["WB_INDEX",1],["WB_REQ_SET_EXTERN",[["WB_EXTERN_DATA_LEN", 6],["WB_EXTERN_DATA", [0, 11 , 0, 0, 0, 0]]]]]]]'

Falls ich sonst noch was liefern kann, lass es mich wissen.

Peter

git-kick, 2025-06-01:
<stale issue>

closed 2025-06-01 by git-kick