git-kick/ioBroker.e3dc-rscp #216: Phase 1 und 2 werden nicht übermittelt

Celle1, 2024-05-06:
Hallo,

für Phase 0 werden alle relevanten Daten übermittelt. Bei Phase 1 und 2 aber nur die Spannung in Volt. Alle anderen Werte fehlen. Im Online-Monitor hingegen werden alle angezeigt.

git-kick, 2024-05-06:
Hallo, danke für die Nachricht.
Kannst du bitte die vollständigen ioBroker Objektnamen nennen, um die es geht.
Was ist mit "Online-Monitor" gemeint?
Wahrscheinlich brauche ich auch ein Debug-Log, wenn es ein Fehler im Adapter ist - bitte ab Adapter-Start 1-2 Minuten aufzeichnen und das Debug-Log hier als Datei posten.

Celle1, 2024-05-06:
Hallo,
es sieht so aus, als ob der EMS diese Daten verwaltet und diese bisher nicht "abgefragt" werden?!  Das müssten diese Daten hier sein. 
[image]

Ich hatte erst gedacht, dass diese über den Wechselrichter (PVI) kommen müssten, hier werden aber nur die Spannungen übermittelt. 

Vielleicht kannst du sie ja noch einbinden. 

Grüße

Celle1, 2024-05-06:
ich meinte schon den AC Kreis: 

Hier werden aber keine Daten bei Phase 1 und 2 übermittelt. Nur die Voltanzeige:
[image]

Celle1, 2024-05-06:
TAG_PVI_REQ_AC_POWER  ist auf S eingestellt. 

Auch eine Neuinstallation führt zu keinem neuen Erkenntnissen.

git-kick, 2024-05-06:
> ich meinte schon den AC Kreis:
> 
> Hier werden aber keine Daten bei Phase 1 und 2 übermittelt. Nur die Voltanzeige: ![grafik](https://private-user-images.githubusercontent.com/139883281/328265064-3fcc5f5b-c1d9-43ba-b553-1911ad0198da.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUwMjgzNDcsIm5iZiI6MTcxNTAyODA0NywicGF0aCI6Ii8xMzk4ODMyODEvMzI4MjY1MDY0LTNmY2M1ZjViLWMxZDktNDNiYS1iNTUzLTE5MTFhZDAxOThkYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwNlQyMDQwNDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05ZDQzZDc1MjEzYzExZTg2YThlN2FkOTE4YzY3MGQ5OTlhMDg5MjI5YmQ5YmI0YzgzNTZlNmQ1MDc4NTI5NTZkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.JkOYT5pROLgQxr7Uo_qrDD7Ud7Uu0ZVfoTrBbQ9RlFg)

Was du siehst, ist mit hoher Wahrscheinlichkeit das, was deine Anlage via RSCP meldet. Die Rest-Unsicherheit könnte ich evtl. mit einem Debug-Log beseitigen, s.o.

Celle1, 2024-05-07:
[iobroker.2024-05-07.log](https://github.com/git-kick/ioBroker.e3dc-rscp/files/15236883/iobroker.2024-05-07.log)
vielleicht hilft das ja

git-kick, 2024-05-10:
Danke für das logfile. Ich sehe darin eine Menge Meldungen zum `tagCode=0x100003e`, dieses wird in Issue #211 behandelt.

Dann fällt mir auf, dass zwei Instanzen von e3dc-rscp laufen (.0 und .1) - ist das Absicht? Wenn das zwei E3/DC Anlagen sind, sollten sie sich eigentlich nicht in die Quere kommen, aber das Szenario habe ich bisher nicht gesehen. Sollten beide Instanzen auf die selbe IP-Adresse/Port gehen, ist das wahrscheinlich problematisch.

Zu eigentlichen Issue kann ich nichts erkennen. Wenn du ein Silly-log einstellst, könnte ich weiter analysieren. Bisher habe ich keinen Anhaltspunkt, dass im Adapter etwas schief läuft.

Celle1, 2024-05-10:
[iobroker.2024-05-10.log](https://github.com/git-kick/ioBroker.e3dc-rscp/files/15276954/iobroker.2024-05-10.log)
Hier der weitere Log-file. 

mit dem Programm: E3/DC RSCPGui kann die benötigen Daten einsehen. Also müssten Sie doch auch bei deinem Tool einsehbar sein, oder?

[image]

git-kick, 2024-05-10:
Danke, schau ich mir an.
Der Analogieschluss gilt nur, wenn RSCPGui nicht auch Werte über Modbus liest (was ich nicht weiß).
RSCP ist leider eine proprietäre und ziemlich fragile Schnittstelle, aber die einzige zum "schreiben".

git-kick, 2024-05-10:
Hier der betreffende Auszug aus dem logfile (kommt sehr oft analog vor):

```
2024-05-10 08:29:30.587  - [35msilly[39m: e3dc-rscp.0 (744) IN: magic: >E3DC< is OK  - ctrl: >0011< is OK - Version 1, with CRC  - seconds: 1715354971 - nseconds: 221238000 - length: 682
TAG_PVI_DATA - type: 0x0E - Container - length: 675 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 0 
TAG_PVI_DC_MAX_STRING_COUNT - type: 0x03 - UChar8 - length: 1 value: 0x02 
TAG_PVI_STATE - type: 0x0D - CString - length: 10 value: 0x103f131d 
TAG_PVI_AC_POWER - type: 0x0E - Container - length: 20 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 0 
TAG_PVI_VALUE - type: 0x0A - Float32 - length: 4 value: 1427 
TAG_PVI_AC_VOLTAGE - type: 0x0E - Container - length: 20 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 0 
TAG_PVI_VALUE - type: 0x0A - Float32 - length: 4 value: 236.89999389648438 
TAG_PVI_AC_CURRENT - type: 0x0E - Container - length: 20 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 0 
TAG_PVI_VALUE - type: 0x0A - Float32 - length: 4 value: 6.050000190734863 
TAG_PVI_AC_APPARENTPOWER - type: 0x0E - Container - length: 20 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 0 
TAG_PVI_VALUE - type: 0x0A - Float32 - length: 4 value: 1439 
TAG_PVI_AC_ENERGY_ALL - type: 0x0E - Container - length: 24 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 0 
TAG_PVI_VALUE - type: 0x0B - Double64 - length: 8 value: 27977273 
TAG_PVI_AC_POWER - type: 0x0E - Container - length: 20 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 1 
TAG_PVI_VALUE - type: 0x0A - Float32 - length: 4 value: 0 
TAG_PVI_AC_VOLTAGE - type: 0x0E - Container - length: 20 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 1 
TAG_PVI_VALUE - type: 0x0A - Float32 - length: 4 value: 237.5 
TAG_PVI_AC_CURRENT - type: 0x0E - Container - length: 20 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 1 
TAG_PVI_VALUE - type: 0x0A - Float32 - length: 4 value: 0 
TAG_PVI_AC_APPARENTPOWER - type: 0x0E - Container - length: 20 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 1 
TAG_PVI_VALUE - type: 0x0A - Float32 - length: 4 value: 0 
TAG_PVI_AC_ENERGY_ALL - type: 0x0E - Container - length: 24 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 1 
TAG_PVI_VALUE - type: 0x0B - Double64 - length: 8 value: 0 
TAG_PVI_AC_POWER - type: 0x0E - Container - length: 20 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 2 
TAG_PVI_VALUE - type: 0x0A - Float32 - length: 4 value: 0 
TAG_PVI_AC_VOLTAGE - type: 0x0E - Container - length: 20 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 2 
TAG_PVI_VALUE - type: 0x0A - Float32 - length: 4 value: 236.6999969482422 
TAG_PVI_AC_CURRENT - type: 0x0E - Container - length: 20 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 2 
TAG_PVI_VALUE - type: 0x0A - Float32 - length: 4 value: 0 
TAG_PVI_AC_APPARENTPOWER - type: 0x0E - Container - length: 20 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 2 
TAG_PVI_VALUE - type: 0x0A - Float32 - length: 4 value: 0 
TAG_PVI_AC_ENERGY_ALL - type: 0x0E - Container - length: 24 <Container content follows...> 
TAG_PVI_INDEX - type: 0x05 - UInt16 - length: 2 value: 2 
TAG_PVI_VALUE - type: 0x0B - Double64 - length: 8 value: 0 
```

Hier sieht man genau, welche Werte der Adapter von der E3/DC erhält:

Für Index 0 sind die Werte plausibel:
- TAG_PVI_AC_POWER = 1427
- TAG_PVI_AC_VOLTAGE = 236.89999389648438
- TAG_PVI_AC_CURRENT = 6.050000190734863
- etc.

Für Index 1 (und ebenso 2) dagegen kommen nur Nullwerte, ausgenommen AC_VOLTAGE:
- TAG_PVI_AC_POWER = 0
- TAG_PVI_AC_VOLTAGE = 237.5
- TAG_PVI_AC_CURRENT = 0
- etc.

Ich sehe aktuell keinen Weg, andere Werte über RSCP zu bekommen. Ein Workaround über Modbus wäre meines Erachtens an der falschen Stelle angesetzt, zumal es offensichtlich bei den meisten Anlagen dieses Problem nicht gibt, hier z.B. die Werte meiner E3/DC S10:

[image]

Celle1, 2024-05-10:
Danke für deine Erklärungen. 

ja, sehr komisch. Naja - muss ich wohl einen weiteren Smartmeter nutzen. 

LG

closed 2024-05-10 by Celle1

git-kick, 2024-05-11:
> Hallo, es sieht so aus, als ob der EMS diese Daten verwaltet und diese bisher nicht "abgefragt" werden?! Das müssten diese Daten hier sein. ![Screenshot 2024-05-06 165332](https://private-user-images.githubusercontent.com/139883281/328226842-2f18d7d1-afc1-4fd5-b9da-a1d6415071b0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU0MjQwMTAsIm5iZiI6MTcxNTQyMzcxMCwicGF0aCI6Ii8xMzk4ODMyODEvMzI4MjI2ODQyLTJmMThkN2QxLWFmYzEtNGZkNS1iOWRhLWExZDY0MTUwNzFiMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxMVQxMDM1MTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMGJjOTA1MDU5OTNhYTQxZGIyYWUzYzcxNWRjZTNkZjA5MjJkNTM1MTdmZjg5YTdjYTMxMmMzMDg0M2RlMDVmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.MIiJVxSIhUZohEyx_HD-O6hEcI1WkZHYm0pF7da43wk)
> 
> Ich hatte erst gedacht, dass diese über den Wechselrichter (PVI) kommen müssten, hier werden aber nur die Spannungen übermittelt.
> 
> Vielleicht kannst du sie ja noch einbinden.
> 
> Grüße
> 
> PS: Online-Monitor, das Portal von E3DC.

Ich habe mir das jetzt auch im aktuellen RSCPGui angesehen. Der Screenshot von dir @Celle1 stammt aus dem Tab "Leistungsmesser", korrekt?

Die oben diskutierten Werte werden aber im Tab "Wechselrichter" auf der rechten Seite  als "Messdaten Phasenbezogen (L1-L3)" gezeigt:
[image]

Was wird denn bei dir im Tab "Wechselrichter" angezeigt?

Celle1, 2024-05-11:
Hallo,
ja genau. Wechselrichter werden nur die Daten für die 1. Phase komplett und bei Phase 2 und 3 nur die Volt Zahl angezeigt.

Im Power Meter dann aber alles. 

Danke fürs "aufnehmen" auf die todo Liste. 

grüße