git-kick/ioBroker.e3dc-rscp #220: Update stable version in repo from 1.2.5 to 1.2.6

GermanBluefox, 2024-06-09:
# Think about update stable version to 1.2.6
**Version**: stable=**1.2.5** (204 days old) => latest=**1.2.6** (15 days old)
**Installs**: stable=**772** (71.95%), latest=**63** (5.87%), total=**1073**

Click to use [developer portal](https://www.iobroker.dev/adapter/e3dc-rscp/releases)
Click to [edit](https://github.com/ioBroker/ioBroker.repositories/edit/master/sources-dist-stable.json#L515)

**Do not close this issue manually as a new issue will be created if condition for update still exists.**

Please drop a comment if any reason exists which blocks updating to version 1.2.6 at this time.

Note: This is an automatically generated message and not personally authored by bluefox!
      @mcm1957 for evidence

mcm1957, 2024-06-21:
Is there any reason to delay updating stable any longer?
Please drop a commit if you know any reason to refrain from updating.

If there is no response until 05.07.2024 we could decide to update anyway.

git-kick, 2024-06-24:
I tried to set v1.2.6 stable using
https://www.iobroker.dev/adapter/e3dc-rscp/releases/~to-stable/1.2.6
but got an error message
`[E605] No actual year found in copyright. Please add "Copyright (c) 2024 Ulrich Kick <iobroker@kick-web.de>" at the end of README.md`

So I updated 2023 to 2024 in the README.md copyright notice, and committed to the master branch.
Can I make this retrofit known to the adapter check, without creating a new version?

git-kick, 2024-06-27:
Done using direct edit.
[E605] will be fixed in e3dc-rscp v1.3.0

Thanks for your support.

mcm1957, 2024-07-07:
Please fix issues reported by adapetr checker.
Errors must be fixed and will require a new release.
Please check which warnings could be fixed easily although warnings will not be blocking.

git-kick, 2024-07-08:
Now I have fixed the errors and most of the warnings
a) in branch #223 (based on V1.2.6)
b) in master (v1.3.0)

How is the best way to release - do I have to skip v1.2.6 and release v1.3.0 now? Will v1.2.6 then disappear in iobroker.dev? 
(I know how to deprecate it in npmjs.com)

mcm1957, 2024-07-08:
Simply release a nee version as usual
Olderbersion will not disaporar - but thats not problem. Ist vompletely normal that some releases only aporar as latest and neuer go to stable.

closed 2024-07-15 by GermanBluefox