git-kick/ioBroker.e3dc-rscp #222: Please fix issues reported by adapter checker

mcm1957, 2024-07-07:
❗ [E026] "{'engines': {'node'>='16'}}" is required at [package.json](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/package.json), "{'engines':{'node'>='18'}}" is recommened
 ❗ [E033] @iobroker/adapter-core 2.5.1 specified. 3.1.4 is required as minimum, 3.1.6 is recommended. Please update dependencies at [package.json](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/package.json)
 ❗ [E036] @iobroker/testing 2.5.4 specified. 4.1.3 is required as minimum, 4.1.3 is recommended. Please update devDependencies at [package.json](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/package.json)
 ❗ [E182] Please remove "common.license" from [io-package.json](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/io-package.json) as "common.licenseInformation" is declared.
 👀 [W181] "common.license" in [io-package.json](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/io-package.json) is deprecated. Please define object "common.licenseInformation"
 👀 [W184] "common.main" is deprecated and ignored. Please remove from [io-package.json](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/io-package.json). Use "main" at [package.json](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/package.json) instead.
 👀 [W184] "common.materialize" is deprecated for admin >= 5 at [io-package.json](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/io-package.json). Please use property "adminUI".
 👀 [W202] Version of [package.json](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/package.json) (1.3.0) doesn't match latest version on [NPM](https://www.npmjs.com/package/iobroker.e3dc-rscp) (1.2.6)
 👀 [W521] "gulp" found at devDependencies at [package.json](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/package.json) but no "gulpfile.js" found. Is this dependency really required?
 👀 [W522] Please consider migrating to admin 5 UI (jsonConfig).

git-kick, 2024-07-08:
Thank you for the list of issues - most of them seem trivial, but: I failed to fix some them for hours because the github action (test-and-release.yml) and the adapter checker seem to have different, inconsistent criteria, e.g.
- test-and-release.yml requires common.license
- adapter checker says common.license has to be removed

I am not sure how to keep the package-files up-to-date, perhaps you have a hint for me how to continuously maintain those files.

mcm1957, 2024-07-08:
Only the outdated test requires common.license. Please Update testing to current Release and checks should be in sync.

mcm1957, 2024-08-05:
replaced by #224

closed 2024-08-19 by git-kick