git-kick/ioBroker.e3dc-rscp #225: Frage Adaptereinstellung SET_POWER Wiederholintervall

ArnoD15, 2024-08-14:
Hallo Uli,

bei mir häufen sich die Fälle, wo bei einem Update vom Adapter e3dc-rscp bei den Instanzeinstellungen vom Adapter bei `SET_POWER Wiederholintervall` ein Wert > 0 eingetragen wird.
[https://forum.iobroker.net/topic/32976/e3dc-hauskraftwerk-steuern/2617](url)

Das führt in Verbindung mit meinem Script dazu, dass der letzte Wert ständig wiederholt wird und somit das Laden oder Entladen der Batterie verhindert wird.

Kann man das irgendwie absichern, dass standardmäßig 0 eingetragen wird? 
Ich habe bereits versucht, eine Möglichkeit zu finden diese Instanzeinstellung vom Adapter über Javascript abzufragen, leider habe ich da nichts gefunden.

VG Arno

git-kick, 2024-08-14:
Hallo Arno,

ich bin nicht sicher, was Du mit "Update" meinst. Installation eines Adapter-Updates oder neuer Werte, die über den Adapter kommen?

Was ich zu `setpower_interval` sagen kann:
1. Der Default ist 10 (Sekunden), siehe [io-package.json](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/io-package.json)
2. Der Wert kann im Config-GUI manuell geändert werden, siehe [index_m.html](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/admin/index_m.html)
3. Ansonsten wird der Wert _ausschließlich_ gelesen, siehe insbesondere [main.js](https://github.com/git-kick/ioBroker.e3dc-rscp/blob/master/main.js)

D.h. ich kann mir nicht erklären, wie (außer im Config-GUI) der Wert von `setpower_interval` geändert wird.
Aber vielleicht habe ich auch deine Frage nicht ganz verstanden.

Zum Nachtrag: `e3dc-rscp.0.EMS.SET_POWER_VALUE` ist der Wert in Watt, mit dem geladen werden soll. Das ist etwas anderes als das Zeitintervall `setpower_interval` (Sekunden). Da verstehe ich Deinen Ansatz nicht.

ArnoD15, 2024-08-14:
Hallo Uli,

anscheinend kann es vorkommen, dass bei Installation eines Adapter-Updates bei SET_POWER Wiederholintervall der Default Wert eingetragen wird und nicht 0.
Was mich jetzt etwas verwirrt ist das der Default Wert 10 ist und nicht 1.
Neuer Link:
[https://forum.iobroker.net/topic/32976/e3dc-hauskraftwerk-steuern/2617](https://forum.iobroker.net/topic/32976/e3dc-hauskraftwerk-steuern/2617)

Es kann aber auch sein, dass der Adapter neu installiert wurde und dann der Default Wert eingetragen wurde und anschließend übersehen wurde diesen zu korrigieren.
Kann man den Default Wert auf 0 ändern? 
Bei Default 0 kann erstmal nichts Negatives passieren, bei 10 kann die Batterie komplett entladen werden, ohne dass man es bemerkt, wenn vom Script keine neuen Werte übermittelt werden.

>Zum Nachtrag: e3dc-rscp.0.EMS.SET_POWER_VALUE ist der Wert in Watt, mit dem geladen werden soll. Das ist etwas anderes als das Zeitintervall setpower_interval (Sekunden). Da verstehe ich Deinen Ansatz nicht.

Durch die Einstellung SET_POWER Wiederholintervall >0 schickt der e3dc-rscp Adapter im eingestellten Wiederholintervall den Wert `e3dc-rscp.0.EMS.SET_POWER_VALUE` an die E3DC Steuerung.
So wird abgesichert, dass die Gegenstelle noch aktiv ist und neue Werte übermittelt.
Wenn jetzt aber das ganze von einem Script gesteuert werden soll, muss natürlich das vom Skript erledigt werden, sonst ist diese Absicherung ausgehebelt.
Genau das passiert aber, wenn jemand bei den Instanzeinstellungen vergisst `SET_POWER Wiederholintervall`auf 0 zu setzen bzw. übersieht den Default Wert zu ändern.

Hoffe, das einigermaßen verständlich erklärt zu haben :-)

git-kick, 2024-08-15:
> anscheinend kann es vorkommen, dass bei Installation eines Adapter-Updates bei SET_POWER Wiederholintervall der Default Wert eingetragen wird und nicht 0.

Meines Wissens werden bei Installation eines Adapter-Updates alle Einstellungen auf Default gesetzt und man kann sie aus einem *.json-file (das man in der letzten Version heruntergeladen hat) wiederherstellen. D.h. nach einer Adapter-Installation muss man immer den gewünschten Wert setzen, sofern man nicht den Default (10) verwenden will.

> Was mich jetzt etwas verwirrt ist das der Default Wert 10 ist und nicht 1.

Also den Wert 1 setzt mein Code nicht, da bin ich so sicher, wie man als Entwickler halt sein kann...

> Kann man den Default Wert auf 0 ändern?
> Bei Default 0 kann erstmal nichts Negatives passieren, bei 10 kann die Batterie komplett entladen werden, ohne dass man es bemerkt, wenn vom Script keine neuen Werte übermittelt werden.

Der Default (10) ist bewusst so gewählt, damit ein (nicht programmierender) ioBroker-Nutzer, der `SET_POWER` ändert, einen nachhaltigen Effekt sieht und nichts über das Intervall wissen muss. Ich befürchte dass es viele Issues geben wird, wenn ich den auf 0 setze. Im übrigen wäre es ja auch gar keine Lösung, wenn er bei Dir auf 1 Sekunde steht und du weißt nicht warum...

> Nachtrag: Habe gerade versucht das Problem zu lösen, indem ich als letzten Wert vom Script bei e3dc-rscp.0.EMS.SET_POWER_VALUE immer null eintrage, aber das wir auch als 0 interpretiert.

> Durch die Einstellung SET_POWER Wiederholintervall >0 schickt der e3dc-rscp Adapter im eingestellten Wiederholintervall den Wert e3dc-rscp.0.EMS.SET_POWER_VALUE an die E3DC Steuerung.
> So wird [...]

Hmm, ich verstehs vielleicht immer noch nicht: wenn du "als letzten Wert vom Script bei `e3dc-rscp.0.EMS.SET_POWER_VALUE` immer null" einträgst, dann werden 0 Watt eingestellt, und zwar im Default alle 10 Sekunden (vorausgesetzt, `SET_POWER_MODE` steht entsprechend, da gehe ich mal davon aus). Kannst du in deinem Code nicht den `setpower_interval` Konfigwert in der e3dc-rscp-Instanz auf das gewünschte Intervall setzen? Dann erledigt das Wiederholen der Adapter für dich. Oder du schreibst in deinen Code eine Zeile, die "immer wieder sicherheitshalber" `setpower_interval = 0` setzt, um diese Funktion des Adapters abzuschalten.
Bin aber gerade nicht sicher, ob und wie das geht, dass man bei einem Adapter "von außen" die Konfig ändert, ich denke sie müsste erreichbar sein unter `system.adapter.e3dc-rscp.native.config` - zur Not müsste ich im Adapter ein Hilfs-Objekt einführen, das den Wert von `setpower_interval` spiegelt, also z.B. `e3dc-rscp.0.EMS.SET_POWER_INTERVAL` - das wäre aber unschön, eigentlich ist die Adapter-Konfig ja dazu da, solche Werte zu halten.

Was meinst? :-)

ArnoD15, 2024-08-16:
Manchmal hat man einfach ein Brett vorm Kopf :-).
`SET_POWER_MODE` ist natürlich die Lösung.
Mein Script war so aufgebaut, dass einfach keine neuen Werte gesetzt wurden, wenn E3DC die Regelung wieder übernehmen sollte.
Das hatte aber auch zur Folge, dass die letzten Werte natürlich nicht geändert wurden und durch das Wiederholintervall (wenn man vergessen hat dieses auf 0 zu setzen) weiter übertragen wurden. Wenn man aber `SET_POWER_MODE` auf 0 setzt, wird ´SET_POWER_VALUE` ignoriert. 
Ich muss also nur mein Script anpassen, das immer, wenn E3DC die Regelung übernehmen soll `SET_POWER_MODE` auf 0 gesetzt wird.

Sorry, dass ich dich damit belästigt habe, aber ohne deinen Denkanstoß mit `SET_POWER_MODE` hätte ich wahrscheinlich ewig nach einer Lösung gesucht.
Danke für deine Hilfe und das Issue kann natürlich geschlossen werden. :-)

closed 2024-08-16 by ArnoD15

git-kick, 2024-08-16:
Sehr gerne Arno!
Ich bin mir ja auch oft nicht sicher ob meine Annahmen stimmen - RSCP ist etwas obskur ;-)