git-kick/ioBroker.e3dc-rscp #230: [JsonConfig] Cannot read file "jsonConfig.json: Not exists

ArnoD15, 2024-09-09:
Hallo Uli,

hast du bei der neuen Version 1.3.0 noch eine Datei vergessen?
Ich habe den Adapter komplett deinstalliert und die Version 1.3.0 neu installiert, jetzt kann ich aber die Instanz Einstellungen nicht mehr öffnen.
Unten im Fenster erscheint die Fehlermeldung [JsonConfig] Cannot read file "jsonConfig.json: Not exists.

VG Arno

git-kick, 2024-09-09:
Hallo Arno, wieder der Schnellste!
Die Datei ist vorhanden, sie wird aber weggefiltert, da musste ich noch die json5 Extension in package. json ergänzen.
Wenn du mit URL (master branch) installierst, funktioniert es.
v1.3.1 schiebe ich asap nach.
VG, Uli

P.S. warum das Problem in meiner Entwicklungsumgebung nicht auftrat, weiß ich nicht...

ArnoD15, 2024-09-09:
Ok, dann versuche ich mal master branch :-)

Danke für die schnelle Antwort.

git-kick, 2024-09-10:
v1.3.1 ist auf npm, und jetzt installierts auch vollständig.

closed 2024-09-10 by git-kick