git-kick/ioBroker.e3dc-rscp #236: Mit der neuen Version kann man 2 Sperrzeiten einstellen und die Funktion Laden/Entladen einzeln einstellen

SurfGargano, 2024-10-03:
**Describe the bug**  
Mit der neuen Version kann man 2 Sperrzeiten einstellen und die Funktion Laden/Entladen bei beiden wählen.
In den Objects ist aber nur eine Sperrzeit jeweils für Laden und Entladen.

Bitte ändern in 2 Sperrzeiten mit Funktionsauswahl Laden/Entladen.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '....'
4. See error

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

_In most cases, a debug-log of adapter startup (about 30 secs) will be helpful or neccessary._

**Versions:**  
 - E3/DC: \<model and battery type\>
 - Adapter version: \<adapter-version\>
 - JS-Controller version: \<js-controller-version\> _determine this with `iobroker -v` on the console_
 - Node version: \<node-version\> _determine this with `node -v` on the console_
 - Operating system: \<os-name and version\>

**Additional context**  
Add any other context about the problem here.

git-kick, 2024-10-03:
Ja, E3/DC hat eine komplett neue Mechanik für die Idle-Periods (zusätzlich) eingeführt. Folgende neue Tags habe ich bis jetzt gefunden:

- TAG_EMS_REQ_SET_IDLE_PERIODS_ENABLE
- TAG_EMS_IDLE_PERIOD_2
- TAG_EMS_SET_IDLE_PERIODS_2
- TAG_EMS_GET_IDLE_PERIODS_2
- TAG_EMS_SET_IDLE_PERIODS_ENABLE
- TAG_EMS_GET_IDLE_PERIODS_ENABLE
- TAG_EMS_PERIOD_ACTIVE
- TAG_EMS_PERIOD_WEEKDAYS
- TAG_EMS_PERIOD_START
- TAG_EMS_PERIOD_STOP

Wird eingebaut, Zeitpunkt jedoch ungewiss.

SurfGargano, 2024-10-03:
OK, Danke

git-kick, 2025-02-28:
Jetzt ist eine erste Implementierung da: [v1.4.0](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/v1.4.0)
Vielleicht magst du das mal testen?

SurfGargano, 2025-03-01:
Das Schreiben der Wochentage funktioniert nicht richtig .
Eingestellt auf der E3DC Webseite Mo-So. 
Gelesen in den Objekten 1234567.  Das ist richtig.
Geändert in den Objekten : 123456
In der Ed3Dc Webseite steht dann Mo,Di, Fri.
Rückgelesen in den Objekten 125, was wiederum der Anzeige in E3DC entspricht.
Ich habe auch andere Einstellungen der Wochentage probiert, sieht ziemlich willkürlich aus.

Außerdem wird noch beim Adapterstart mehrmals e3dc-rscp.0.EMS.IDLE_PERIODS_DISCHARGE in den Objekten erzeugt und wieder gelöscht.

Zeiten setzen, Enable/Disable und Period Type funktionieren

git-kick, 2025-03-01:
Okay, danke!

Das mit den Wochentagen ist komisch, hat bei mir funktioniert dachte ich - muss ich genauer reinschauen.

Die alten IDLE_PERIODS_(DIS)CHARGE werden tatsächlich jedel Mal im Adapter gelöscht und neu aufgebaut, wenn von der E3/DC die idle periods (V1 oder V2) kommen - das ist für mich die "Wahrheit". Kurz nach dem Adapterstart passiert das ein paarmal, weil ich mehrere Anfragen lossschicke. Danach sollte es sich beruhigen.

Ich habe auch schon überlegt, ob ich in den Adaptereinstellungen die Auswahl "nur V1" oder "nur V2" einbaue, denn beide zu verwenden führt wahrscheinlich zu Inkonsistenzen. Was denkst du dazu?

SurfGargano, 2025-03-01:
> Okay, danke!
>
> Das mit den Wochentagen ist komisch, hat bei mir funktioniert dachte ich - muss ich genauer reinschauen.
>
> Die alten IDLE_PERIODS_(DIS)CHARGE werden tatsächlich jedel Mal im Adapter [...]

Das mit der V1 und V2 finde ich gut. Evtl. als Hover eine Beschreibung dazu.

git-kick, 2025-03-01:
Also, ich hatte tatsächlich im letzten Moment noch einen Fehler in die `weekdayStringToBitmask()` Funktion eingebaut. Das ist jetzt im master branch behoben.

Das v1.4.0 pre-release werde ich nochmal plattmachen und v1.4.0 neu (mit der Korrektur) generieren. Vielleicht baue ich zuvor auch noch die entweder V1 oder V2 Wahl ein.

closed 2025-03-01 by git-kick