git-kick/ioBroker.e3dc-rscp #241: Powermeter Module zeigt nur einen Leistungsmesser

CRachfahl, 2024-10-07:
Zuerst einmal super das Ihr das PM ergänzt habt.

Ich habe aktuell 2 Leistungsmesser und bekomme noch einen dritten. Es wird leider bei mir neue Version 3.1 gerade aus dem beta reposetory installiert nur ein PM angezeigt.

Ebenso hat er die tauchen in dem Abfrageinterwallen die PM werte nicht auf. Es gibt auch eine Meldung zu den PM werten

Gruss Carsten

[image]

[image]

git-kick, 2024-10-08:
> 
> Ich habe aktuell 2 Leistungsmesser und bekomme noch einen dritten. Es wird leider bei mir neue Version 3.1 gerade aus dem beta reposetory installiert nur ein PM angezeigt.

Der Adapter fragt (analog wie bei den Batterien) in aufsteigender Reihenfolge PMs ab, bis er keine mehr findet.
Offenbar sind aber die PMs nicht lückenlos durchnummeriert. (Da ich selbst nur einen PM habe, konnte ich das nicht testen.)
Ich kann gelegentlich versuchen, das "blind" anzupassen, wenn du bereit bist zu testen.
 
> 
> Ebenso hat er die tauchen in dem Abfrageinterwallen die PM werte nicht auf. Es gibt auch eine Meldung zu den PM werten
> 
Bitte eine ganz neue Adapterkonfiguration anlegen, siehe [README]([image] - dann sollten die PM-Tags in der Tabelle für die Abfrageintervalle auftauchen.

CRachfahl, 2024-10-08:
Hi Uli,

super gerne teste ich!

Zur Erklärung: ich habe bei mir jetzt drei Solaranlagen und komme damit am Netztzugangspunkt über 30kW. Deswegen schreibt mein Netzbetreiber eine Wandler Messung vor. Hierzu brauchst du einen externen Leistungsmesser von E3DC der hat die ID 6. Weiterhin habe ich noch einen weiteren für die alten Solarwechselrichter, der hat dann die ID 2.

Lege heute Abend einen neuen Adapter an.

Danke Carsten

Gerne teste ich.

Bitte eine ganz neue Adapterkonfiguration anlegen, siehe [README]([grafik]<https://github.com/git-kick/ioBroker.e3dc-rscp?tab=readme-ov-file#reuse-of-adapter-configuration> - dann sollten die PM-Tags in der Tabelle für die Abfrageintervalle auftauchen.

CRachfahl, 2024-10-20:
Hi Uli,

habe jetzt wie besprochen die drei Leistungsmesser mit der Nummer 0, 2, 6. 

Ich habe nun in der main.js des Adapters testweise die Zeile 474 "this.maxIndex["PM"] = 1" auf 6 gesetzt und er erzeugt mir als Ergebnis unter PM die drei Powermeter.

Gruß Carsten

[image]

git-kick, 2024-10-22:
@CRachfahl, ja genau, kann man so machen. Vermutlich hast du viele Warnungen im log, weil deine Anlage sagt: PM1,3,4,5 kenn ich nicht.

Mein Anspruch für den Adapter ist, dass er erkennt, welche PMs es gibt und nur diese abfragt.

CRachfahl, 2024-10-22:
Hi Uli,

klar geht das besser! Ich warte auch schon gespannt auf deine Version. Brauchte nur zur Visualisierung schon mal die Daten aus den Leistungsmessern.

Werde gerne testen wenn du was hast.

Gruß Carsten

git-kick, 2024-10-25:
Hallo Carsten,

hier ist ein Branch zum Testen (basiert auf v1.3.1):
[Issue#241](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23241)
Wenn du das startest, sollte im Log bald nach dem Instance-Start folgendes stehen:
`(info) Adjusted PM index set to [0,2,6]`
und zuvor einige ähnliche Zeilen, wo die Menge noch größer ist, da sortiert der Adapter die aus, wo ein Fehler zurückkommt.

Bin gespannt ob es auf Anhieb klappt!

CRachfahl, 2024-10-25:
> Hallo Carsten,
>
> hier ist ein Branch zum Testen (basiert auf v1.3.1): [Issue#241](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23241) Wenn du das startest, sollte im Log bald nach dem [...]

Hi Uli,

teste ich gerne sofort. Eine blode Frage habe ich allerdings: wie installiere ich den Branch auf ioBroker? 

Carsten

git-kick, 2024-10-25:
[image]

In "Adapter" auf die Katze, dann "Benutzerdefiniert", dann bei "URL" eingeben:
`https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23241`

CRachfahl, 2024-10-25:
Hi Uli,

klappt super!!!! Hier ein Auszug aus dem Log. Es sind auch die drei Leistungmesser angelegt.

Klasse bin happy!

Wenn du jemand zum Testen auf dem S20 X Pro braucht bin dabei.

Carsten

[image]
[image]

git-kick, 2024-10-25:
Danke fürs schnelle Testen!
Du kannst den Branch an sich verwenden, das war seit 1.3.1 die erste Änderung.
Bis 1.3.2 offiziell da ist, wird es noch dauern...

git-kick, 2025-03-01:
Kommt mit v1.4.1 (Release heute).

closed 2025-03-01 by git-kick