git-kick/ioBroker.e3dc-rscp #243: Warning/error with v. 1.3.1 and E3DC firmware 2024_024 and _028

blue-bean, 2024-10-22:
**Describe the bug**  
A clear and concise description of what the bug is.

**To Reproduce**  
Steps to reproduce the behavior:
start the adapter and wating

**Expected behavior**  
Works without errors "failed twriting data to socket"

**Screenshots & Logfiles**  
[iobroker.2024-10-22.txt](https://github.com/user-attachments/files/17475988/iobroker.2024-10-22.txt)

_In most cases, a debug-log of adapter startup (about 30 secs) will be helpful or neccessary._

**Versions:**  
 - E3/DC: S10 E Pro
 - Adapter version: 1.3.1
 - JS-Controller version: 8.8.3
 - Node version: 20.18.0
 - Operating system: docker container on unraid 6.12.13

**Additional context**  
Add any other context about the problem here.

git-kick, 2024-10-22:
Thank you for the report.
At a first glance, this looks like a config or LAN issue.
For further investigation, please provide a debug log covering e3dc-rscp instance startup plus ca. 15 secs.

blue-bean, 2024-10-22:
Here is the debug log
[debuglog.2024-10-22.txt](https://github.com/user-attachments/files/17480317/debuglog.2024-10-22.txt)

git-kick, 2024-10-23:
I cannot see the instance startup.
Please give me the timestamp when e3dc-rscp was started, and the debug log covering startup and subsequent 15 secs.

blue-bean, 2024-10-23:
Here again. Is it correct? If not please tell me what should i do in detail.
[debug.2024-10-23.txt](https://github.com/user-attachments/files/17493647/debug.2024-10-23.txt)

adpater instance starts at 2024-10-23 16:39:43.942

git-kick, 2024-10-23:
Thank you. The debug.log is perfect now.

My observation is that there is a problem with large inbound messages (from E3DC to ioBroker). 

Small messages like TAG_INFO_SERIAL_NUMBER (at 2024-10-23 16:40:06.718) are ok and processed by the adapter. 

But larger messages (eg 
2024-10-23 16:40:06.725) have consistent length or wrong magic/ctrl code. These are indicators that something broke at TCP level.

We saw such before when the LAN connection has short interrupts, and usually the adapter recovers gently. In your case the ratio of broken large messages seems so high that results received by the adapter do not make sense anymore.

In short, I am quite sure this is some network issue on TCP level or below, not an adapter issue.

git-kick, 2024-11-04:
@blue-bean, thank for following up. Always good to have an explanation for _strange_ behavior...

closed 2024-11-04 by git-kick