git-kick/ioBroker.e3dc-rscp #247: Change the "price cap for charging the home power plant batter."

minnegg3, 2025-02-01:
I'm looking for a possibilty to change the price cap for charging the home power plant battery = Preisgrenze für das Laden der Hauskraftwerksbatterie via ioBroker.

Would be great to get this feature :-;
I want to set the price cap on a daily basis by the minimum price of tibber so the batterie of e3dc will always chage at the time of the mimimum price per day.

git-kick, 2025-02-01:
Hi,
yes, I saw some new tags concerning Price Based Charging. But there is no RSCP interface documentation about the new features (afaik), so it will take some time to extend the adapter.
Please also see #237

git-kick, 2025-06-02:
Sorry, took some time. 

If you like, you can test this [branch](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23247)
The new objects are `EMS.DPP...`

Will be released (later) with v1.4.3

git-kick, 2025-09-21:
Released with v1.4.3

closed 2025-09-21 by git-kick