git-kick/ioBroker.e3dc-rscp #248: Some values are no longer updated

pglidingmatth, 2025-02-01:
**Describe the bug**  
The values TAG_BAT_REQ_RSOC_REAL and TAG_BAT_REQ_RSOC are no longer updated automatically. For an update I have to restart the instance

**Expected behavior**  
I have set the instance to update the values after 5 minutes at the latest. However, they no longer update at all.

**Screenshots & Logfiles**  

[image]

[image]

[image]

**Versions:**  
 - E3/DC: S10 X
 - Adapter version: 1.3.1

git-kick, 2025-02-01:
Thank you for the report.

As the values are _no longer_ updated, it may be some adapter installation/update issue. Did you update the e3dc-rscp adapter recently?

In any case, for analysis I need a debug log (pls post as attachment, not inline). It should cover instance startup plus 30 secs.

pglidingmatth, 2025-02-02:
Thank you very much for the extremely quick response. Enclosed you will find the log. I hope I have made it available to you correctly here.

[iobroker.current.log](https://github.com/user-attachments/files/18631685/iobroker.current.log)

pglidingmatth, 2025-02-02:
Regarding your question, yes I did update to version v1.3.1 some time ago. It may be that the problem has existed for a little longer and I didn't notice it.

git-kick, 2025-02-02:
Thank you for the log. I can't see anything special about TAG_BAT_REQ_DATA or the corresponding response.

Could you provide a silly-log?

git-kick, 2025-02-02:
off topic: if you want to use PM values and avoid "has no assigned polling interval" messages, you have to re-new the adapter config, see [reuse of adapter configuration](https://github.com/git-kick/ioBroker.e3dc-rscp?tab=readme-ov-file#reuse-of-adapter-configuration)

git-kick, 2025-02-03:
I digged into the code an found that RSOC / RSOC_REAL polling intervals are assigned by different tags (due to RSCP anomalies):

- TAG_BAT_REQ_RSOC_REAL is scheduled by TAG_BAT_REQ_RSOC_REAL = 'S'
- TAG_BAT_REQ_RSOC is scheduled by TAG_BAT_REQ_INFO ='?' (not in your screenshots)

So it might be that the two attributes are updated differently - perhaps you can observe and this will help help finding the issue.

- How is TAG_BAT_REQ_INFO set in your polling intervals?
- Are both RSOC and RSOC_REAL "never" updated unless you restart the e3dc-rscp instance?

closed 2025-03-01 by git-kick