git-kick/ioBroker.e3dc-rscp #249: Nur zwei Bateriemodule werden angezeigt

CRachfahl, 2025-02-08:
Bei meinem S20 X Pro mit vier 21 kW Bateriemodulen werden nur die ersten beiden angezeigt. denke könnte eine Begrenzung der ausgewerteten Bateriemodule sein.

Hier die Übersicht des Biobroker Modules:

[image]

Hier die Ansicht im E3DC Portal:
[image]

Wenn Tests erforderlich sind bin gerne bereit diese zu tun.

Gruß CRachfahl

git-kick, 2025-02-08:
Ja, das ist korrekt. Dies ist der erste mir bekannte Fall mit mehr als 2 BATs.
Das ist etwas kurios, weil in der E3/DC Tagliste in einem Kommentar steht, dass der BAT_INDEX immer 0 ist - der Kommentar ist wohl veraltet.

Jedenfalls kannst Du das einfach in main.js ändern, etwa in Zeile 473:
`this.maxIndex["BAT"] = 1;`
Die 1 steht für max. 2 BATs (0 und 1). Bei dir wäre also mindestens 3 einzutragen.

Werde ich mit v1.3.2 hochsetzen.

closed 2025-02-08 by git-kick

CRachfahl, 2025-02-08:
Super werde ich testen.

P.S.: das S 20 X Pro kam an zwei Bateriestrangen sogar je 4 Bateriemodulwe also theoretisch 8.