git-kick/ioBroker.e3dc-rscp #262: Wallbox charging data not written into data points

github-ps1304, 2025-03-14:
**Describe the bug**  
After update to 04.2 the Wallbox data (charging) is not written into data points

**To Reproduce**  
Steps to reproduce the behavior:
1. electric car is charging
2. no data visible under objects of e3dc-rscp - WB

**Expected behavior**  
A clear and concise description of what you expected to happen.

charging data like WB_All, WB_Grid or WB_Sun should be written into data points

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

[image]

[image]

_In most cases, a debug-log of adapter startup (about 30 secs) will be helpful or neccessary._

**Versions:**  
 - E3/DC: \S10E and LG Batteries
 - Adapter version: \0.4.2\>
 - JS-Controller version: \7.0.6
 - Node version: \20.18.3
 - Operating system: \debian on proxmox\>

**Additional context**  
Add any other context about the problem here.

P-B-aus-R, 2025-03-14:
Hallo,
bei mir das gleiche Problem. Seit Version 1.4.1 werden alle Objekte unterhalb von WB nicht mehr aktualisiert. Es bleiben die letzten Werte stehen.
Mit der Vorversion wurden die Werte von WB_0 und WB_1 noch aktualisiert.
Grüße,
Peter

git-kick, 2025-03-15:
Hallo,
da ich keine E3/DC-Wallbox habe, kann ich das nicht reproduzieren.
Ich bitte um ein Debug-Logfile (ca. 1 Minute ab Adapter-Start).

P-B-aus-R, 2025-03-15:
Hallo,
ich habe zum ersten mal versucht ein Log abzuspeichern. Die htm-Datei, die dort raus kam, ließ sich hier nicht hochladen, habe deshalb eine Zip draus gemacht. Ich hoffe dass es so funktioniert

[iobroker.current.zip](https://github.com/user-attachments/files/19265088/iobroker.current.zip)

.
Danke.
Peter

github-ps1304, 2025-03-17:
hier mein LogFile

[E3DC-rscp-debug.txt](https://github.com/user-attachments/files/19280735/E3DC-rscp-debug.txt)

git-kick, 2025-03-17:
Danke für die log-files. Ich sehe, dass die Anfragen zu den spezifischen Wallbox-Daten nicht mehr rausgehen. Aber ich kann noch nicht recht nachvollziehen, weshalb. 

Ich habe jetzt die Initialisierung für die Wallbox-Anzahl geändert, weil das evtl. den Aufruf verhindert hat. Vielleicht könnt Ihr das testen, Installation von [Issue#262](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23262)

Mir ist aufgefallen, dass Ihr anscheinend nicht die neueste Adapter-Konfiguration habt; bitte den Adapter komplett neu installieren und die Konfiguration _nicht_  aus einer *.json laden. (siehe [Reuse of adapter configuration](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/master?tab=readme-ov-file#reuse-of-adapter-configuration))

github-ps1304, 2025-03-17:
Habe auf dem Test System den Adapter + Instanz gelöscht und neu installiert.

[E3DC-rscp-debug-newinstalled.txt](https://github.com/user-attachments/files/19284898/E3DC-rscp-debug-newinstalled.txt)

git-kick, 2025-03-17:
An der Konfig-Seite sehe ich, dass _nicht_ der Branch [Issue#262](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23262) lief. Denn da sieht die Konfig-Seite so aus:

[image]

(siehe Feld "Wallbox" rechts unten)

Wenn du es nochmal probierst, dann wähle bitte logging "silly/alles" und schränke auf den Adapter e3dc-rscp ein. Der Zeitraum soll nur einen Start plus ca. 1 min umfassen.

github-ps1304, 2025-03-17:
Danke - habe jetzt die Branch installiert; werde morgen wieder das Auto laden - Objekte sind schon mal alle da.

P-B-aus-R, 2025-03-17:
Ich hoffe, ich habe alles richtig gemacht, da ich hier am Grenzbereich dessen bin, wass mein Wissen mit dem iobroker hergibt. Die Logdatei ist anbei. 
Die Wallbox-Daten scheinen mit der Version wieder zu funktionieren. Die zweite Wallbox habe ich nicht getestet, da die von der Steuerung sowieso noch nicht richtig funktioniert hat. Vielen Dank. Wäre schön wenn irgendwann die zweite Wallbox noch gesteuert werden könnte.

[iobroker-log.txt](https://github.com/user-attachments/files/19298021/iobroker-log.txt)

github-ps1304, 2025-03-18:
sieht gut aus - Datenobjekte werden wie erwartet mit Daten gefüllt !! Vielen Dank.

anbei das Log.

[E3DC-rscp-debug-262-silly.txt](https://github.com/user-attachments/files/19315703/E3DC-rscp-debug-262-silly.txt)

closed 2025-03-20 by git-kick