git-kick/ioBroker.e3dc-rscp #263: Steuerung des Wertes für die Notstromreserve geht nicht mehr

Steffel99, 2025-03-15:
Ich habe bisher problemlos die Notstromreserve meines S10E über e3dc-rscp.0.EP.PARAM_EP_RESERVE setzen können. Das funktioniert seit dem Update auf 1.4.1 leider nicht mehr. Offenbar ist ist das jetzt read only. Ich habe versucht, das im Objekt zu ändern und den write Zugriff wieder herzustellen, aber auch das geht nicht.

Im iobroker Fehlerlog steht nach dem Versuch die Einstellung zu ändern:

Read-only state "e3dc-rscp.0.EP.PARAM_EP_RESERVE" has been written without ack-flag with value "4"
wobei der Value 4 das ist, was ich testweise eingegeben habe.

Ist das ein Bug oder Absicht, dass das setzen nicht mehr funktioniert? Habe darüber nichts gefunden. Alles andere funktioniert weitherhin fehlerfrei.

git-kick, 2025-03-16:
Danke für den Bericht.
Der beobachtete Effekt ist nicht beabsichtigt.
Ich werde versuchen, das zu reproduzieren.

git-kick, 2025-03-17:
Habe den Fehler gefunden, das sollte jetzt wieder wie gehabt funktionieren.
Ist im master behoben: https://github.com/git-kick/ioBroker.e3dc-rscp

Release mit v1.4.2 (kann noch etwas dauern).
Du kannst so lange auch v1.3.1 verwenden.

closed 2025-03-17 by git-kick