git-kick/ioBroker.e3dc-rscp #270: DB: HistoryDataDay.SUM.DC_Power wird nicht gelesen

Lucifor1976, 2025-06-06:
Hallo

Die DB Werte History Data Day werden nicht (mehr) ausgelesen.
Die Werte für Woche und Monat kommen, die für den Tag nicht.
Das ganze seit ich den Adapter auf 1.4.2 gebracht habe.
Ein Reroll auf Version 1.3.1 oder 1.2.6 brachte keinen Erfolg.
Es ging aber und ich habe seit Monaten mit diesem Datenpunkt einen schönen Char anlegen können.

Ich glaube ich versehe die ganze Datenbank von e3dc nicht und kann der Struktur der Datenbank so garnicht folgen.
Die Dev Readme hab ich gelsen, werde aber nicht schlau daraus.

Jemand eine Idee?

**Versions:**  
 - E3/DC: S10E
 - Adapter version: 1.4.2
 - JS-Controller version: 7.0.6
 - Node version: 20.19.2
 - Operating system: Debian 12 Bookworm

git-kick, 2025-06-06:
Hallo, das klingt strange.

Kannst du mir ein debug-log geben mit dem erfolglosen Versuch (DAY) und einem erfolgreichen Lauf - können auch zwei separate logs sein. Bitte immer mit beginnend mit dem Adapterstart.

Vielleicht finde ich dann was; kann aber nichts versprechen...

Lucifor1976, 2025-06-06:
Hier mal ein log, ziemlich viele Fehler.

Der Timestamp von den Datenpunkten in week hat heute 0 Uhr.
Der Timestamp vom Day (alle) ausser DC_Power ist heute 4:00 Uhr.
derTimestamp von Day DC_Power ist gestern abend 20:45, sowohl gestern alsauch heute wert = 0.

`2025-06-06 11:16:16.229 - info: host.iobroker stopInstance system.adapter.e3dc-rscp.0 (force=false, process=true)
2025-06-06 11:16:16.232 - info: e3dc-rscp.0 (508723) Got terminate signal TERMINATE_YOURSELF
2025-06-06 11:16:16.235 - info: e3dc-rscp.0 (508723) terminating
2025-06-06 11:16:16.236 - info: e3dc-rscp.0 (508723) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2025-06-06 11:16:16.237 - warn: e3dc-rscp.0 (508723) E3/DC connection closed
2025-06-06 11:16:16.237 - info: e3dc-rscp.0 (508723) Stop communication with E3/DC and pause a minute before retry ...
2025-06-06 11:16:16.336 - info: host.iobroker stopInstance system.adapter.e3dc-rscp.0 send kill signal
2025-06-06 11:16:16.738 - info: e3dc-rscp.0 (508723) terminating
2025-06-06 11:16:16.869 - info: host.iobroker instance system.adapter.e3dc-rscp.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
2025-06-06 11:16:20.281 - info: host.iobroker instance system.adapter.e3dc-rscp.0 in version "1.4.2" started with pid 619680
2025-06-06 11:16:21.572 - info: e3dc-rscp.0 (619680) starting. Version 1.4.2 in /opt/iobroker/node_modules/iobroker.e3dc-rscp, node: v20.19.2, js-controller: 7.0.6
2025-06-06 11:16:21.759 - info: e3dc-rscp.0 (619680) Connection to E3/DC is established
2025-06-06 11:16:21.833 - info: e3dc-rscp.0 (619680) Adjusted PM index set to [0,2,3,4,5,6,7,8,9]
2025-06-06 11:16:21.833 - info: e3dc-rscp.0 (619680) Adjusted PM index set to [0,3,4,5,6,7,8,9]
2025-06-06 11:16:21.834 - info: e3dc-rscp.0 (619680) Adjusted PM index set to [0,4,5,6,7,8,9]
2025-06-06 11:16:21.834 - info: e3dc-rscp.0 (619680) Adjusted PM index set to [0,5,6,7,8,9]
2025-06-06 11:16:21.834 - info: e3dc-rscp.0 (619680) Adjusted PM index set to [0,6,7,8,9]
2025-06-06 11:16:21.834 - info: e3dc-rscp.0 (619680) Adjusted PM index set to [0,7,8,9]
2025-06-06 11:16:21.835 - info: e3dc-rscp.0 (619680) Adjusted PM index set to [0,8,9]
2025-06-06 11:16:21.835 - info: e3dc-rscp.0 (619680) Adjusted PM index set to [0,9]
2025-06-06 11:16:21.835 - info: e3dc-rscp.0 (619680) Adjusted PM index set to [0]
2025-06-06 11:16:21.858 - info: e3dc-rscp.0 (619680) Decreased BAT max. index to 4
2025-06-06 11:16:21.871 - info: e3dc-rscp.0 (619680) Decreased BAT max. index to 3
2025-06-06 11:16:21.876 - info: e3dc-rscp.0 (619680) Decreased BAT max. index to 2
2025-06-06 11:16:21.881 - info: e3dc-rscp.0 (619680) Decreased BAT max. index to 1
2025-06-06 11:16:21.895 - info: e3dc-rscp.0 (619680) Device reports that PVI.REQ_VOLTAGE_MONITORING is not available - setting polling interval to 'N'
2025-06-06 11:16:21.895 - info: e3dc-rscp.0 (619680) Device reports that PVI.REQ_FREQUENCY_UNDER_OVER is not available - setting polling interval to 'N'
2025-06-06 11:16:21.902 - info: e3dc-rscp.0 (619680) Decreased PVI max. index to 0
2025-06-06 11:16:22.191 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_STATUS (0xe000004)
2025-06-06 11:16:22.191 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_ERROR_CODE (0xe000005)
2025-06-06 11:16:22.192 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_MODE (0xe000006)
2025-06-06 11:16:22.192 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_DEVICE_ID (0xe00000b)
2025-06-06 11:16:22.192 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_DEVICE_NAME (0xe000042)
2025-06-06 11:16:22.192 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_DEVICE_STATE (0xe060000)
2025-06-06 11:16:22.192 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_APP_SOFTWARE (0xe000007)
2025-06-06 11:16:22.192 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_BOOTLOADER_SOFTWARE (0xe000008)
2025-06-06 11:16:22.192 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_HW_VERSION (0xe000009)
2025-06-06 11:16:22.192 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_FLASH_VERSION (0xe00000a)
2025-06-06 11:16:22.192 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_ENERGY_ALL (0xe000001)
2025-06-06 11:16:22.192 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_ENERGY_SOLAR (0xe000002)
2025-06-06 11:16:22.192 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_SOC (0xe000003)
2025-06-06 11:16:22.193 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_PM_POWER_L1 (0xe00000c)
2025-06-06 11:16:22.193 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_PM_POWER_L2 (0xe00000d)
2025-06-06 11:16:22.193 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_PM_POWER_L3 (0xe00000e)
[... cut ...]
2025-06-06 11:31:21.943 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_PM_POWER_L3 (0xe00000e)
2025-06-06 11:31:51.806 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_PM_POWER_L1 (0xe00000c)
2025-06-06 11:31:51.806 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_PM_POWER_L2 (0xe00000d)
2025-06-06 11:31:51.806 - warn: e3dc-rscp.0 (619680) Received data type ERROR: RSCP_ERR_NOT_AVAILABLE (6) - tag TAG_WB_REQ_PM_POWER_L3 (0xe00000e)
`

git-kick, 2025-06-06:
Bitte ein debug-log: die Instanz auf 'debug' umstellen (statt 'info').
Hast du eine Wallbox an der E3/DC? Sieht nicht so aus. Schalte mal WB und andere nicht genutzte Namespaces aus.
Die Logs bitte als File anhängen, nicht inline.

Lucifor1976, 2025-06-06:
Doch es ist eine Wallbox angeschlossen, die steht gerade im Sonnenmodus und läd mein Auto.

Hier das Debug und ich hoffe ich finde das FIle anhängen :)

[log.txt](https://github.com/user-attachments/files/20626513/log.txt)

Lucifor1976, 2025-06-06:
Und hier der Rest des log, auch nach 5 Minuten wird Week und Day nicht nachgeliefert....

[log.txt](https://github.com/user-attachments/files/20627043/log.txt)

Lucifor1976, 2025-06-06:
Und hier eine silly Version...

[log.txt](https://github.com/user-attachments/files/20627189/log.txt)

git-kick, 2025-06-06:
Also das silly-log sieht gut aus, hier der Ausschnitt:  [history_data_day.log](https://github.com/user-attachments/files/20627654/history_data_day.log)

Was mich irritiert, ist die letzte Zeile:
`2025-06-06 13:30:11.561 - silly: e3dc-rscp.0 (717818) processTree: path = , tree = [ , , , , , , , , , , , , ]>, , , , , , , , , , , , , ]>, , , , , ...`
Eigentlich zeigt das silly-log die ganze Datenstruktur mit Inhalt. Hast du gefiltert, wegen Datenschutz? (wäre verständlich)

Ich versuche auf jeden Fall noch, das bei mir zu reproduzieren.

Lucifor1976, 2025-06-06:
Iobroker Protokoll, Silly, und dann herunterladen.

Gibts woanders noch was besseres?

git-kick, 2025-06-06:
> 
> Edit: Wenn das aus deinem Log ist, ist bei dir die Zeile ebenfalls mit 0 eingetragen.. Schau mal die Zeile: TAG_DB_DC_POWER - type: 0x0A - Float32 - length: 4 value: 0

Geht es nur um den Wert von TAG_DB_DC_POWER, alles andere kommt auch bei Dir?

Lucifor1976, 2025-06-06:
Das letzte DB Firmwareupdate ist vom 21.03.2025
Das letzte Update insgesamt vom 08.05.

Es lief alles bis zum Update, ich habe ihne komplett neu eingerichtet und nichts aus einem Backup genommen.

git-kick, 2025-06-06:
Also, ich verwende selbst DB nicht, deshalb kann ich nicht einmal sagen, ob die TAG_DB_DC_POWER Werte jemals ok waren; aber dich habe ich so verstanden, dass das schon mal i.O. war? Das ist jedenfalls fast sicher out-of-my-scope, weil ich keinen Draht zu E3/DC habe (also die Firmware nicht debuggen oder ändern kann).

Das Thema "komplett Week und Day wird nicht gemeldet" ist vermutlich ein anderes. Und es ist sporadisch, weil es ja vorhin (siehe history_data.log) bei Dir auch mal funktioniert hat. Das riecht nach timing-Problem, sehr schwer zu finden. Im Moment steh ich auf dem Schlauch, aber ich denk nochmal nach...

git-kick, 2025-06-06:
> Das letzte DB Firmwareupdate ist vom 21.03.2025 Das letzte Update insgesamt vom 08.05.

Was ist das "DB Firmwareupdate"? meinst du die E3/DC Firmware?

Was meinst du genau mit "Das letzte Update insgesamt"? E3/DC Firmware oder ioBroker oder e3dc-rscp-Adapter?

> 
> Es lief alles bis zum Update, ich habe ihne komplett neu eingerichtet und nichts aus einem Backup genommen.

Hier geht es jetzt um den Adapter, richtig? Heisst das: es lief alles o.k., bis zu zum ersten Mal e3dc-rscp v1.4.2 installiert hast, und seitdem sind beide Probleme da?

git-kick, 2025-06-06:
Noch eine Idee: hast du die Möglichkeit, eine ioBroker-Test-Instanz neu einzurichten und es dort zu probieren? Das wäre ein guter Hinweis zu wissen, ob dort das selbe Problem besteht - dann könnte man zumindest eingrenzen zwischen deine E3/DC, dein Netzwerk, oder ... 

Ich tu mir halt schwer, wenn bei mir alles geht :-\

git-kick, 2025-06-06:
> Jaein.. Mit Daten von vorgestern, es werden immer die gleichen Daten übertragen und daher nicht in den Datnpunken geschrieben (weil keine Änderung)

Ah! Das ist wichtig: der Adapter bringt einen Wert zurück (z.B. immer 0), aber der wird nicht in den Datenpunkt geschrieben, weil da schon 0 steht. Wenn dies die Auffälligkeit ist, dann hier zwei Statements dazu:

1. (oben schon gesagt) wen die Anlage Mist meldet, zeigt der Adapter Mist

2. Der e3dc-rscp-Adadpter hat eine Einstellung `setState()-Aufruf nur bei Änderung` - wenn du die abwählst, dann wird der Datenpunkt auch aktualisiert, wenn der gleiche Wert gemeldet wurde.

Das erklärt aber immer noch nicht, warum du das Problem genau seit dem v1.4.2 update hast...

git-kick, 2025-06-06:
> .. tuts ja nicht, auch bei dir stimmt der Inhalt des Wertes garantiert nicht, ich glaube nicht das Deine Anlage gerade aus ist, somit kann der Wert nicht null sein, auch bei dir nicht.. ist er aber..

Das sehe ich anders: mein Fokus ist auf dem Adapter (weil ich den beeinflussen kann), und der läuft bei mir ohne Auffälligkeit. Dass das RSCP-Interface bzw. die Firmware Fehler, das "ist halt so", _bullshit-in-bullshit-out_.

Für mich ist diese "Spitzfindigkeit" extrem wichtig, sonst kämpfe ich hier gegen Windmühlen.

Lucifor1976, 2025-06-06:
Version 1.4.1

History_Data_Year.SUM.DC_Power = ja
(Alle Datenpunkte in YEAR)
History_Data_Month.SUM.DC_Power = ja
(Alle Datenpunkte in MONTH)
History_Data_Week.SUM.DC_Power = ja
(Alle Datenpunkte in WEEK)
History_Data_Day.SUM.DC_Power = nein
(Alle Datenpunkte in DAY, Alle Datenpunkt beinhalten Daten vom 04.06.2025 22:30 Uhr)

Obacht, da isser! findet aber den Weg nicht in die Datenpunkte!

`2025-06-06 17:45:21.454  - [35msilly[39m: e3dc-rscp.2 (921519) IN: magic: >E3DC< is OK  - ctrl: >0011< is OK - Version 1, with CRC  - seconds: 1749224723 - nseconds: 288562000 - length: 6607
TAG_DB_HISTORY_DATA_WEEK - type: 0x0E - Container - length: 6600 <Container content follows...> 
TAG_DB_SUM_CONTAINER - type: 0x0E - Container - length: 143 <Container content follows...> 
TAG_DB_GRAPH_INDEX - type: 0x0A - Float32 - length: 4 value: -1 
TAG_DB_BAT_POWER_IN - type: 0x0A - Float32 - length: 4 value: 100555.5 
TAG_DB_BAT_POWER_OUT - type: 0x0A - Float32 - length: 4 value: 95475 
TAG_DB_DC_POWER - type: 0x0A - Float32 - length: 4 value: 343064 `

Zeitpunkt 06.06.2025 17:45 Uhr

Lucifor1976, 2025-06-06:
Mit Version 1.4.2 dann wieder das gleich wie heute früh.

`2025-06-06 18:15:29.253  - [35msilly[39m: e3dc-rscp.2 (944632) IN: magic: >E3DC< is OK  - ctrl: >0011< is OK - Version 1, with CRC  - seconds: 1749226531 - nseconds: 140387000 - length: 3907
TAG_DB_HISTORY_DATA_DAY - type: 0x0E - Container - length: 3900 <Container content follows...> 
TAG_DB_SUM_CONTAINER - type: 0x0E - Container - length: 143 <Container content follows...> 
TAG_DB_GRAPH_INDEX - type: 0x0A - Float32 - length: 4 value: -1 
TAG_DB_BAT_POWER_IN - type: 0x0A - Float32 - length: 4 value: 0 
TAG_DB_BAT_POWER_OUT - type: 0x0A - Float32 - length: 4 value: 6717 
TAG_DB_DC_POWER - type: 0x0A - Float32 - length: 4 value: 0 `

Jetzt kann ich den Adapter noch auf einen iobroker Slave bringen, dann bin ich am Ende mit meinem Latain...

git-kick, 2025-06-07:
Danke für den ausführlichen Input, ich bin jetzt ein paar Tage unterwegs und kann mich erst danach wieder darum kümmern...

Lucifor1976, 2025-06-09:
Das jetzt n Witz ..

Geht wieder... Ich dreh durch...

Schau mal, wenn der Adapter neu installiert wird, wird der Datenpunkt:
e3dc-rscp.0.db.historydataday.time_span
falsch angelegt. der Wert muss 86400 sein, nicht 21600.

Wert geändert, zack geht wieder....

git-kick, 2025-09-21:
Der erkannte Fehler wurde in Issue https://github.com/git-kick/ioBroker.e3dc-rscp/issues/271 gefixt.
Die übrigen Phänomene sind aktuell nicht nachvollziehbar.

closed 2025-09-21 by git-kick

MatthiasBosch, 2025-11-10:
Hallo,
ich hab auch das Problem, das die History Daten nicht dauerhaft geschrieben werden.
Wenn ich an den Werten "TIME_INTERVAL", "TIME_START", "TIME_SPAN" Änderungen vornehme, dann werden die Daten für ca. 1 Tag geschrieben, danach ist dann allerdings wieder Schluss.

Was muss ich den tun, damit die Daten dauerhaft geschrieben werden?
Soll ich zu bestimmten Zeiten die Werte "TIME_INTERVAL", "TIME_START", "TIME_SPAN" manuell z.B. per Script setzen?

MatthiasBosch, 2025-11-10:
Danke für die schnelle Antwort.

Mir geht es im speziellen z.B. um die Werte "e3dc-rscp.0.DB.HISTORY_DATA_YEAR.SUM.AUTARKY" oder "e3dc-rscp.0.DB.HISTORY_DATA_YEAR.SUM.CONSUMPTION". Wenn ich es nicht übersehen habe, dann bekomme ich diese Werte nur über diese Knoten. Richtig?
Dann bleibt nur, die Werte "TIME_INTERVAL", "TIME_START", "TIME_SPAN" regelmäßig am besten beim Tageswechsel einmal zu setzen. Richtig?