git-kick/ioBroker.e3dc-rscp #271: e3dc-rscp.0.DB.HISTORY_DATA_{DAY,WEEK,MONTH,YEAR}.TIME_{START,INTERVAL,SPAN} wird beim Adapterstart überschrieben

git-kick, 2025-06-13:
**Describe the bug**  
Die genannten Werte werden vom Adapter initialisiert. Das passiert bei jedem Neustart, dabei gehen vom Anwender gesetzte Werte verloren.

**To Reproduce**  
1. Setze e3dc-rscp.0.DB.HISTORY_DATA_DAY.TIME_SPAN auf 86400
2. Starte die Adapterinstanz neu
3. e3dc-rscp.0.DB.HISTORY_DATA_DAY.TIME_SPAN ist wieder auf 21600

**Expected behavior**  
Die im Titel genannten Werte sollen beim Adapterstart nur vorbelegt werden, wenn die DAtenpunkte im Objektbaum  noch nicht vorhanden sind.

**Screenshots & Logfiles**  
Siehe Issue#270

**Versions:**  
 - E3/DC: S10
 - Adapter version: 1.4.2

**Additional context**  
Die ist ein Unterthema von Issue#270 (@Lucifor1976)

git-kick, 2025-06-13:
Implementiert in [Issue#271](https://github.com/git-kick/ioBroker.e3dc-rscp/tree/Issue%23271)
Kommt mit dem nächsten Adapter-Realease.

Lucifor1976, 2025-06-14:
Perfekt, Firma Dankt :)

git-kick, 2025-06-14:
Gerne!

closed 2025-06-14 by git-kick